NVIDIA-Omniverse/PhysX #292: PhysXFoundation_64.dll was not found

mishaBogdanov, 2024-07-05:
Hello, I'm using visual studio 2022 to link to physX. here are all the settings I've changed:

[image]
[image]
[image]
[image]
the .dll seems to be there in my checked folder:
[image]
the code that's giving me issues is

PxDefaultErrorCallback	gErrorCallback;

without it my code compiles. 

all other visual studio settings should be at default, as I reset them all in attempt to fix this issue. my steps to get to this point are 
- download physX
- run generate_projects.bat
- run the visual studio solution that's generated in checked mode
- the snippets work
- start a new project with the above settings.

Any help appreciated, Thanks!!

vreutskyy, 2024-07-05:
Hi @mishaBogdanov 
Did you try to copy PhysX DLLs next to your project's executable? I mean the folder your EXE file is in.

mishaBogdanov, 2024-07-05:
Thank-you so much, this fixed all my issues.

closed 2024-07-05 by mishaBogdanov