NVIDIA-Omniverse/PhysX #294: Are cylinders not supported in PVD?

mishaBogdanov, 2024-07-10:
Hello, I'm trying to create a custom vehicle using physX, and want to use CylinderCallbacks for wheels. however, I can't seem to get to visualize it in PVD, I only get to see the coordinate system, and the velocity vector which seems to resemble the movement of a cylinder, but I'm not sure. Is there any way to visualize cylinders in PVD? here's my code:

```

#include <iostream>
#include "PxPhysics.h"
#include "PxPhysicsAPI.h"
#include "geometry/PxGjkQuery.h"
#include "extensions/PxCustomGeometryExt.h"

using namespace physx;

#define PVD_HOST "127.0.0.1"

static PxDefaultAllocator		gAllocator;
static PxDefaultErrorCallback	gErrorCallback;
static PxFoundation* gFoundation = NULL;
static PxPhysics* gPhysics = NULL;
static PxDefaultCpuDispatcher* gDispatcher = NULL;
static PxScene* gScene = NULL;
static PxMaterial* gMaterial = NULL;
static PxPvd* gPvd = NULL;

static void createCylinderActor(float height, float radius, float margin, const PxTransform& pose)
{
	PxCustomGeometryExt::CylinderCallbacks* cylinder = new PxCustomGeometryExt::CylinderCallbacks(height, radius, 0, margin);
	PxRigidDynamic* actor = gPhysics->createRigidDynamic(pose);
	actor->setActorFlag(PxActorFlag::eVISUALIZATION, true);
	PxShape* shape = PxRigidActorExt::createExclusiveShape(*actor, PxCustomGeometry(*cylinder), *gMaterial);
	shape->setFlag(PxShapeFlag::eVISUALIZATION, true);
	gScene->addActor(*actor);
}

void initPhysics(bool /*interactive*/)
{
	gFoundation = PxCreateFoundation(PX_PHYSICS_VERSION, gAllocator, gErrorCallback);
	gPvd = PxCreatePvd(*gFoundation);
	PxPvdTransport* transport = PxDefaultPvdSocketTransportCreate(PVD_HOST, 5425, 10);
	gPvd->connect(*transport, PxPvdInstrumentationFlag::eALL);
	gPhysics = PxCreatePhysics(PX_PHYSICS_VERSION, *gFoundation, PxTolerancesScale(), true, gPvd);
	//PxInitExtensions(*gPhysics, gPvd);
	PxSceneDesc sceneDesc(gPhysics->getTolerancesScale());
	sceneDesc.gravity = PxVec3(0.0f, -9.81f, 0.0f);
	gDispatcher = PxDefaultCpuDispatcherCreate(2);
	sceneDesc.cpuDispatcher = gDispatcher;
	sceneDesc.filterShader = PxDefaultSimulationFilterShader;
	gScene = gPhysics->createScene(sceneDesc);
	gScene->setVisualizationParameter(PxVisualizationParameter::eSCALE, 1.0f);
	gScene->getScenePvdClient()->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_SCENEQUERIES, true);
	PxPvdSceneClient* pvdClient = gScene->getScenePvdClient();
	if (pvdClient)
	{
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONSTRAINTS, true);
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONTACTS, true);
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_SCENEQUERIES, true);
	}
	gMaterial = gPhysics->createMaterial(0.5f, 0.5f, 0.6f);
	PxRigidStatic* groundPlane = PxCreatePlane(*gPhysics, PxPlane(0, 1, 0, 0), *gMaterial);
	gScene->addActor(*groundPlane);
	createCylinderActor(5.0f, 5.0f, 0.0f, PxTransform(PxVec3(0.0f, 100.0f, 0.0f)));

}

void stepPhysics()
{
	gScene->simulate(1.0f / 60.0f);
	gScene->fetchResults(true);
}

int main()
{
	int i = 0;
	initPhysics(false);
	while (i<10000) {
		stepPhysics();
	}
    std::cout << "Hello World!\n";
}

```

vreutskyy, 2024-07-26:
Hi @mishaBogdanov 
You seem to miss the collision shapes visualization enabling in your sample.
`	gScene->setVisualizationParameter(PxVisualizationParameter::eCOLLISION_SHAPES, 1.0f);
`
Hope this helps

mishaBogdanov, 2024-07-31:
oh, thank-you. sorry, somehow missed the response. this really helps. It's in wireframe mode tho, is there any way to see it as solid? Either way it's not a huge issue.

vreutskyy, 2024-07-31:
no, it's only wireframe as far as I know.

closed 2024-08-13 by mishaBogdanov