NVIDIA-Omniverse/PhysX #296: Blast & Flow compile error

njikmf, 2024-07-17:
### Library and Version

PhysX latest f07ba77f2d8af805824385b4616549974d231a8d

### Operating System

Win10 
### Steps to Trigger Behavior

 run physx/generate_projects.bat  and compile in VS2022 (release),  everything is ok
 
 run blast/build.bat ,  
`
 Populating dependency directory: D:\code\cpp\Physx\blast\_repo\deps\repo_man\omni\repo\man\_vendor
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:997)'))': /artifactory/api/pypi/ct-omniverse-pypi/simple/certifi/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:997)'))': /artifactory/api/pypi/ct-omniverse-pypi/simple/certifi/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:997)'))': /artifactory/api/pypi/ct-omniverse-pypi/simple/certifi/
ERROR: Operation cancelled by user
`

In Flow folder, I found there are "build.bat" and "generate_projects.bat". I don't know which one to run. I tried and they both failed.

```
D:\code\cpp\Physx\flow>build.bat
Sentry is attempting to send 1 pending events
Waiting up to 2 seconds
Press Ctrl-Break to quit
Running packman in preparation for premake ...
Package 'dxc' at version '1.7.2308-win' is missing from local storage.
packman(ERROR): Package not found on specified remote servers! (name: dxc, version: 1.7.2308-win)
Sentry is attempting to send 2 pending events
Waiting up to 2 seconds
Press Ctrl-Break to quit
'\VsWhere.exe' is not recognized as an internal or external command,
operable program or batch file.
'\VsWhere.exe' is not recognized as an internal or external command,
operable program or batch file.
Unsupported Visual Studio version
!!! Failure while building nvflow !!!

D:\code\cpp\Physx\flow>generate_projects.bat
The system cannot find the path specified.
```

At last, try to compile the "nvfloweditor"
```
1>------ Build started: Project: nvfloweditor, Configuration: release x64 ------
1>ImguiRenderer.cpp
1>D:\code\cpp\Physx\flow\source\nvfloweditor\ImguiRenderer.cpp(36,10): error C1083: Cannot open include file: 'shaders/ImguiCS.hlsl.h': No such file or directory
```

### Code Snippet to Reproduce Behavior

```
<your-snippet>
```

### Expected Behavior

### Actual Behavior

gamerx1221, 2024-07-18:
also having this issue

also would be helpful if packman repo has list enabled??  currently packman says cloudfront_uri is not setup to provide a list

cnem255, 2024-08-08:
I also have this issue. I've tried a plethora of things, but it all seems to lead to more and more problems.