NVIDIA-Omniverse/PhysX #315: In the PxCooking stage, are collinear triangles removed?

szhang21, 2024-09-14:
In the PhysX 5.1.3 documentation, it is mentioned: "Mesh cleaning removes invalid triangles (containing out-of-range vertex references), duplicate triangles, and zero-area triangles." However, the baked triangle data still includes collinear triangles.

szhang21, 2024-09-14:
I am sure I did not set PxMeshPreprocessingFlag::eDISABLECLEANMESH.

closed 2024-09-14 by szhang21

vreutskyy, 2024-09-16:
Hi @szhang21 
Thanks for the report, we'll check this. It would help if you could provide some sample/data to reproduce the issue.

szhang21, 2024-09-18:
Hello, I have checked, and this issue has been resolved in the latest version of PhysX. In PhysX 5.1.3, the condition being checked is area2==0.0f. The value 0.0f is not sufficient to ensure that the area of the transformed triangle remains 0, leading to the assert failure in disconnection.Thank you for your reply