NVIDIA-Omniverse/PhysX #326: Crash afer overflowing maxSolverFrictionProgress (PxU16)

ikkah, 2024-11-08:
### Library and Version

v5.4.2-106.1 (latest available)

Also reproduced with 5.3.1.

### Operating System

Windows 10

### Steps to Trigger Behavior

Unfortunately I don't have a test case I could share, but the issue occurs in situations where more than 2^16 contacts are generated.

Articulations are not used in the scene.

### Expected Behavior

I'd expect PhysX to error or warn without crashing, possibly omitting the contacts over 2^16.

### Actual Behavior

There is an overflow of maxSolverFrictionProgress in recordStaticConstraint() (DyConstraintPartition.cpp). 

PhysX eventually crashes with the following call stack:

```
 	PhysX_64.dll!physx::Dy::`anonymous namespace'::writeConstraintDesc<physx::Dy::`anonymous namespace'::RigidBodyClassification>(const physx::PxSolverConstraintDesc * descs, unsigned int numConstraints, physx::Dy::`anonymous-namespace'::RigidBodyClassification & classification, physx::PxArray<unsigned int,physx::PxReflectionAllocator<unsigned int>> & accumulatedConstraintsPerPartition, physx::PxSolverConstraintDesc * eaTempConstraintDescriptors, physx::PxSolverConstraintDesc * eaOrderedConstraintDesc, unsigned int maxPartitions, unsigned int numOverflows) Line 615	C++
>	PhysX_64.dll!physx::Dy::batchConstraints<0,physx::Dy::`anonymous namespace'::RigidBodyClassification>(const physx::PxSolverConstraintDesc * eaConstraintDescriptors, unsigned int numConstraintDescriptors, physx::Dy::`anonymous-namespace'::RigidBodyClassification & classification, physx::PxArray<unsigned int,physx::PxReflectionAllocator<unsigned int>> & constraintsPerPartition, physx::PxSolverConstraintDesc * eaOverflowConstraintDescriptors, unsigned int maxPartitions, physx::PxSolverConstraintDesc * eaOrderedConstraintDescriptors, unsigned int & numOverflows, unsigned int & numOrderedConstraints, unsigned int & numStaticConstraints) Line 845	C++
 	PhysX_64.dll!physx::Dy::partitionContactConstraints(physx::Dy::ConstraintPartitionOut & out, const physx::Dy::ConstraintPartitionIn & in) Line 878	C++
 	PhysX_64.dll!physx::Dy::PartitionTask::runInternal() Line 2254	C++
```

I couldn't quite figure out the connection between the overflow and the subsequent crash, but the crash seems to happen every time after the overflow.

Please let me know if you need more information.

vreutskyy, 2024-11-08:
Hi @ikkah, thanks for reporting.

ikkah, 2024-11-11:
Looks like I can reproduce the crash with this modified version of SnippetContactReportCCD ("checked" build config, MSVC).

It's a slightly ridiculous setup with a million triangle meshes, but not that different from the actual use-case I have, to be honest.

```
#include <vector>
#include "PxPhysicsAPI.h"
#include "../snippetutils/SnippetUtils.h"
#include "../snippetcommon/SnippetPrint.h"
#include "../snippetcommon/SnippetPVD.h"

using namespace physx;

static PxDefaultAllocator		gAllocator;
static PxDefaultErrorCallback	gErrorCallback;
static PxFoundation*			gFoundation			= NULL;
static PxPhysics*				gPhysics			= NULL;
static PxDefaultCpuDispatcher*	gDispatcher			= NULL;
static PxScene*					gScene				= NULL;
static PxMaterial*				gMaterial			= NULL;
static PxTriangleMesh*			gTriangleMesh		= NULL;
static PxRigidStatic*			gTriangleMeshActor	= NULL;
static PxRigidDynamic*			gSphereActor		= NULL;
static PxPvd*					gPvd                = NULL;
static PxU32					gSimStepCount		= 0;

std::vector<PxVec3> gContactPositions;
std::vector<PxVec3> gContactImpulses;
std::vector<PxVec3> gContactSphereActorPositions;

static PxFilterFlags contactReportFilterShader(	PxFilterObjectAttributes attributes0, PxFilterData filterData0, 
												PxFilterObjectAttributes attributes1, PxFilterData filterData1,
												PxPairFlags& pairFlags, const void* constantBlock, PxU32 constantBlockSize)
{
	PX_UNUSED(attributes0);
	PX_UNUSED(attributes1);
	PX_UNUSED(filterData0);
	PX_UNUSED(filterData1);
	PX_UNUSED(constantBlockSize);
	PX_UNUSED(constantBlock);

	//
	// Enable CCD for the pair, request contact reports for initial and CCD contacts.
	// Additionally, provide information per contact point and provide the actor
	// pose at the time of contact.
	//

	pairFlags = PxPairFlag::eCONTACT_DEFAULT
			  | PxPairFlag::eDETECT_CCD_CONTACT
			  | PxPairFlag::eNOTIFY_TOUCH_CCD
			  |	PxPairFlag::eNOTIFY_TOUCH_FOUND
			  | PxPairFlag::eNOTIFY_CONTACT_POINTS
			  | PxPairFlag::eCONTACT_EVENT_POSE;
	return PxFilterFlag::eDEFAULT;
}

class ContactReportCallback: public PxSimulationEventCallback
{
	void onConstraintBreak(PxConstraintInfo* constraints, PxU32 count)	{ PX_UNUSED(constraints); PX_UNUSED(count); }
	void onWake(PxActor** actors, PxU32 count)							{ PX_UNUSED(actors); PX_UNUSED(count); }
	void onSleep(PxActor** actors, PxU32 count)							{ PX_UNUSED(actors); PX_UNUSED(count); }
	void onTrigger(PxTriggerPair* pairs, PxU32 count)					{ PX_UNUSED(pairs); PX_UNUSED(count); }
	void onAdvance(const PxRigidBody*const*, const PxTransform*, const PxU32) {}
	void onContact(const PxContactPairHeader& pairHeader, const PxContactPair* pairs, PxU32 nbPairs) 
	{
		std::vector<PxContactPairPoint> contactPoints;

		PxTransform spherePose(PxIdentity);
		PxU32 nextPairIndex = 0xffffffff;

[... 190 lines cut ...]
		PX_RELEASE(gPvd);
		PX_RELEASE(transport);
	}
	PX_RELEASE(gFoundation);

	printf("SnippetContactReportCCD done.\n");
}

int snippetMain(int, const char*const*)
{
#ifdef RENDER_SNIPPET
	extern void renderLoop();
	renderLoop();
#else
	initPhysics(false);

	stepPhysics(false);

	cleanupPhysics(false);
#endif

	return 0;
}

```

PierreTerdiman, 2025-01-09:
I couldn't reproduce this so far. I am getting a crash inside OpenGL from the rendering code first. And if I disable the rendering (undefining RENDER_SNIPPET) then it looks like the snippet finishes successfully.

[image]

[image]

ikkah, 2025-01-09:
Thanks for looking into it, Pierre.

I checked again with latest PhysX and could not reproduce the issue. The snippet worked without crashing for me after just converting the std::vectors into PxArrays.

I also checked again with a clean PhysX 5.4.2 and was able to reproduce the issue with that version, so the issue seems to have been fixed somewhere between those two versions. I'm not yet sure if this solves the crash I had before simplifying it into the snippet, but I'll test later with latest PhysX and reopen the issue if I can still reproduce it.

closed 2025-01-09 by ikkah