NVIDIA-Omniverse/PhysX #331: The SDK build failed using Visual Studio 2022

Space-CN-G, 2024-11-19:
Example: PhysX-106.0-physx-5.4.0

1.I downloaded PhysX SDK 5.4.2 from the Releases page.
2.Opened generate_projects.bat.
3.Selected vc17win64.
4.Used Visual Studio 2022 to open PhysX-106.0-physx-5.4.0\physx\compiler\vc17win64\PhysXSDK.sln.
5.Clicked "Rebuild All."

Here’s the translation of the error log:
`14>LowLevelAABB_static_64.lib(BpAABBManagerBase.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevelAABB_static_64.lib(BpAABBManagerBase.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevelAABB_static_64.lib(BpAABBManager.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevelAABB_static_64.lib(BpAABBManager.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxsNphaseImplementationContext.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxsNphaseImplementationContext.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxsCCD.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxsCCD.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxsSimpleIslandManager.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxsSimpleIslandManager.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxsDefaultMemoryManager.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxsDefaultMemoryManager.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcContactCache.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcContactCache.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxsIslandSim.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxsIslandSim.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcNpBatch.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcNpBatch.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcNpContactPrepShared.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcNpContactPrepShared.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcMaterialMethodImpl.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcMaterialMethodImpl.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcNpCacheStreamPair.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcNpCacheStreamPair.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcNpCacheStreamPair.obj) : error LNK2005: "public: void __cdecl physx::PxcNpCacheStreamPair::reset(void)" (?reset@PxcNpCacheStreamPair@physx@@QEAAXXZ) is already defined in LowLevel_static_64.lib(PxcNpThreadContext.obj)
14>LowLevel_static_64.lib(PxcNpCacheStreamPair.obj) : error LNK2005: "public: __cdecl physx::PxcNpCacheStreamPair::PxcNpCacheStreamPair(class physx::PxcNpMemBlockPool &)" (??0PxcNpCacheStreamPair@physx@@QEAA@AEAVPxcNpMemBlockPool@1@@Z) is already defined in LowLevel_static_64.lib(PxcNpCacheStreamPair.obj)
14>PhysXTask_static_64.lib(TaskManager.obj) : error LNK2038: Detected mismatch in “_ITERATOR_DEBUG_LEVEL”: value “0” does not match value “2” in (NpOmniPvd.obj)
14>PhysXTask_static_64.lib(TaskManager.obj) : error LNK2038: Detected mismatch in “RuntimeLibrary”: value “MD_DynamicRelease” does not match value “MTd_StaticDebug” in (NpOmniPvd.obj)
14>LowLevel_static_64.lib(PxcNpContactPrepShared.obj) : error LNK2005: "void __cdecl combineMaterials(class physx::PxsMaterialManager const *,unsigned short,unsigned short,float &,float &,float &,unsigned int &,float &)" (?combineMaterials@@YAXPEBVPxsMaterialManager@physx@@GGAEAM11AEAI1@Z) is already defined in LowLevel_static_64.lib(PxcNpContactPrepShared.obj)
14>  Creating library C:/Users/Space/Downloads/PhysX-106.0-physx-5.4.0/physx/bin/win.x86_64.vc143.mt/debug/PhysX_64.lib and object C:/Users/Space/Downloads/PhysX-106.0-physx-5.4.0/physx/bin/win.x86_64.vc143.mt/debug/PhysX_64.exp
14>LINK : warning LNK4098: Default library “MSVCRT” conflicts with other libraries; use /NODEFAULTLIB:library
14>LINK : warning LNK4217: Symbol “_fpclass” (defined in “libucrtd.lib(ieeemisc.obj)”) imported by “LowLevel_static_64.lib(PxsCCD.obj)” (function “public: bool __cdecl physx::PxsCCDPair::sweepAdvanceToToi(float,bool)” (?sweepAdvanceToToi@PxsCCDPair@physx@@QEAA_NM_N@Z)”)
`

vreutskyy, 2024-11-19:
Hi @Space-CN-G 
Sorry, but if you downloaded PhysX SDK 5.4.2, why do you open PhysX-106.0-physx-5.4.0\...?
Shouldn't it be PhysX-106.1-physx-5.4.2\...?
Just tried it - builds for me.

closed 2024-11-19 by Space-CN-G

rainstormstudio, 2024-12-29:
for anyone having the same error, here's the solution:

do `vcpkg integrate remove` and build will work; if not, simply uninstall vcpkg from your computer.

I think it's vcpkg's library messing with the build envrionment.