NVIDIA-Omniverse/PhysX #346: [question] function V4ExtractMax seems wrong.

o0olele, 2024-12-14:
~~~c
PX_FORCE_INLINE FloatV V4ExtractMax(const Vec4V a)
{
	const PxF32 t0 = (a.x >= a.y) ? a.x : a.y;
	const PxF32 t1 = (a.z >= a.w) ? a.x : a.w;
	return t0 >= t1 ? t0 : t1;
}
~~~
- is the function [V4ExtractMax](https://github.com/NVIDIA-Omniverse/PhysX/blob/6e8ce6bb3208eb4ab0e4f266284e5b34fc9c3227/physx/include/foundation/PxVecMathAoSScalarInline.h#L1279) return the max value of x,y,z,w in Vec4V? if so, i think is:
~~~c
PX_FORCE_INLINE FloatV V4ExtractMax(const Vec4V a)
{
	const PxF32 t0 = (a.x >= a.y) ? a.x : a.y;
	const PxF32 t1 = (a.z >= a.w) ? a.z : a.w;
	return t0 >= t1 ? t0 : t1;
}
~~~
- same with V4ExtractMin.

preist-nvidia, 2024-12-16:
Thank you @o0olele - that's indeed a copy-pasta bug. We'll fix it.

PierreTerdiman, 2025-05-05:
This has been fixed, and the change will appear in the next release.