NVIDIA-Omniverse/PhysX #349: PxShape doesn't match documentation

HexPerson, 2024-12-30:
### Library and Version

### Operating System

### Steps to Trigger Behavior

 1. Clone repo
 2. Use code according to documentation

### Code Snippet to Reproduce Behavior

```
physx::PxBoxGeometry geom;
_shape->getBoxGeometry(geom);
```

### Expected Behavior

### Actual Behavior

vreutskyy, 2025-01-06:
Hi @HexPerson 

It was removed in 5.2.0. There's a line in CHANGELOG.md
```
* Deprecated passthrough functions in PxShape such as `getGeometryType()` and the specialized `get<geomType>Geometry()` were removed. Calls to these functions should be replaced by the accessing the underlying geometry directly with `getGeometry()`.
```

We'll check the documentation, thanks.

PierreTerdiman, 2025-01-08:
Thank you, we have updated the docs.

closed 2025-01-08 by PierreTerdiman