NVIDIA-Omniverse/PhysX #355: Overlap between convex core and capsule doesn't work

thfProjects, 2025-01-09:
### PhysX 5.5.0

### Windows 11

### Steps to Trigger Behavior

 1. Create a scene with PxCapsuleGeometry shape and a cylinder shape with either PxConvexCoreGeometry or PxCustomGeometry
 2. Do scene overlap query between the two

### Expected Behavior

Detect overlap

### Actual Behavior

No overlap detected, additionally,  PxCustomGeometry with cylinder callbacks would sometimes make the application not respond if it overlaps another shape, but I am unable to find out exactly how to reproduce it

vreutskyy, 2025-01-09:
Hi @thfProjects 
Thanks for the report. It would help if you could provide a code snippet to reproduce this issue.

thfProjects, 2025-01-09:
Hi @vreutskyy, 
This function triggers the bug:

```C++
static void overlapTest(PxPhysics* physics, PxScene* scene, PxMaterial* material) {
	PxCapsuleGeometry capsuleGeometry = PxCapsuleGeometry(1, 1);
	PxConvexCoreGeometry cylinderGeometry = PxConvexCoreGeometry(PxConvexCoreCylinder(1, 1));

	PxActor* capsuleActor = PxCreateKinematic(*physics, PxTransform(PxIdentity), capsuleGeometry, *material, 1);
	PxActor* cylinderActor = PxCreateKinematic(*physics, PxTransform(PxIdentity), cylinderGeometry, *material, 1);

	scene->addActor(*capsuleActor);
	scene->addActor(*cylinderActor);

	int maxOverlaps = 1;

	std::vector<PxOverlapHit> hitBuffer(maxOverlaps);
	PxOverlapBuffer hit(hitBuffer.data(), maxOverlaps);

	if (!scene->overlap(cylinderGeometry, PxTransform(PxIdentity), hit)) {
		printf("No overlaps detected");
	} else {
		printf("Overlaps detected");
	}
}
```

vreutskyy, 2025-01-09:
Thanks!
Could you try to add PxQueryFlag::eANY_HIT flag?

```
	PxOverlapBuffer buf3;
	PxQueryFilterData fd; fd.flags |= PxQueryFlag::eANY_HIT;
	bool res3 = getScene().overlap(PxSphereGeometry(1.0f), PxTransform(PxVec3(-1, 0, 0)), buf3, fd);
	EXPECT_TRUE(res3);
	EXPECT_TRUE(buf3.hasBlock);
```

thfProjects, 2025-01-09:
Tried it, it makes no difference.

vreutskyy, 2025-01-09:
Aa. I didn't realize you're trying to use convex core as a overlap geometry. It's not supported yet.
```
		\param[in] geometry		Geometry of object to check for overlap (supported types are: box, sphere, capsule, convex).
```
`convex` here means convex mesh, not convex core. convex core support will be added later.

thfProjects, 2025-01-09:
Ahh OK. My bad, it didn't occur to me to check if it was supported.
Thank you for clarifying!

vreutskyy, 2025-01-09:
Thanks for finding it. I might forget about that ;)