NVIDIA-Omniverse/PhysX #360: Unclear Blast Authoring extension usage

PavelSupenko, 2025-02-04:
Hello, dear Blast developers!
Maybe I am missing something and this message should be on Discussions tab. If yes, sorry for interruption.
I want to integrate authoring functionality into Unity and already made a lot - I've transferred your SDK to the CMake and created a working bridge for the low-level API. And now I started implementing the required C++ part for authoring (as an example I am using [your deprecated tool from another repository](https://github.com/NVIDIAGameWorks/Blast/blob/dev/tools/AuthoringTool/src/AuthoringTool.cpp)), but found that mentioned in your documentation tool named [ConvexMeshBuilder](https://docs.omniverse.nvidia.com/kit/docs/blast-sdk/latest/docs/api/extensions/ext_authoring.html#convexmeshbuilder) that is required by the `NvBlastExtAuthoringProcessFracture` function is just an abstract class with no inheritors:
```cpp
class ConvexMeshBuilder
{
public:
    virtual void                    release() = 0;
    virtual CollisionHull* buildCollisionGeometry(uint32_t verticesCount, const NvcVec3* vertexData) = 0;
    virtual void releaseCollisionHull(CollisionHull* hull) const = 0;
};
```

In the `NvBlastExtAuthoringProcessFracture` function documentation described that this object should be created by non existed function:
`collisionBuilder    Collision builder created by NvBlastExtAuthoringCreateConvexMeshBuilder`

As I see this collision builder is used inside mentioned function with no null checks or something more so I can't skip it or something like that I think.
Unfortunately I see there are no tests for this functionality so only one information source I can use is the deprecated usage example that has PhysX dependency that is not here anymore.

Could you please help me with the correct flow around this functionality? Maybe I am doing something wrong and I don't need it?

nvidia-bgaldrikian, 2025-02-24:
Hello Pavel,

My apologies for the delay in my reply and for the trouble with the ConvexMeshBuilder interface.  Although the Blast library allows one to leave physics and its associated collision representation to the user, there are certain operations for which the authoring tools  require a convex collision representation, although in many cases there's no need for it to be exact.  The two functions named BlastBondGenerator::createBondBetweenMeshes have a parameter struct BondGenerationConfig, and if you set BondGenerationConfig::bondMode = BondGenMode::Exact then it will compute overlaps between the graphics meshes to determine how the meshes touch and what their bond properties should be.  However if you set that parameter to BondGemMode::AVERAGE then it will use a convex representation, supplied by the user through ConvexMeshBuilder.  Since it's often sufficient to have just a rough approximation of the overlap between two meshes in order to generate a bond that's usable by Blast, you can often get away with something as simple as a bounding box as the return data from ConvexMeshBuilder::buildCollisionGeometry.  In fact, that's what we do in the Omniverse Kit Blast extension.

I hope this is helpful, and it's great to hear that you've made so much progress putting Blast authoring into Unity!

Bryan Galdrikian

PavelSupenko, 2025-02-25:
Thank you, @nvidia-bgaldrikian , I actually didn't expect such a quick response :)

Aha, so for creating bonds I can just calculate the bounding box and use just 8 vertices to represent the mesh. I will try this implementation. Can I keep this issue open for a little time until I finish implementing this functionality?

By the way, I saw that your Omniverse Kit Blast extension does actually what I'm trying to recreate in Unity and It would be great to check this code out. But, unfortunately, I didn't find any open source repositories with that code. Am I right, that Blast extension is closed source as other Omniverse modules and extensions and there are no plans to reveal them publicly?

Pavel Supenko

PavelSupenko, 2025-02-27:
I haven't tested yet the bonds generating, but common fracture logic works on mobile! Tested this on Google Pixel and iPhone and even runtime Voronoi destruction works pretty well (better than I expected)
Prepared a small demo video here: https://youtu.be/MnusdXUDvx0?si=-LPOCKmqkgdRjusH

closed 2025-03-20 by PavelSupenko

nvidia-bgaldrikian, 2025-03-20:
Hi Pavel,

That's really great to see Blast in action on a phone!  I'm glad you were able to get it working, thank you for sharing that.

Regarding the kit extension source goes, however, as far as I know it's not open.  Sorry.

Bryan