NVIDIA-Omniverse/PhysX #364: Reproduce the code of the direct drive vehicle, but find that the vehicle cannot collide with other objects

capp-adocia, 2025-02-11:
## Question
Hello, I am a beginner, and recently I have been working on reproducing the SnippetVehicle2DirectDrive case in PhysX 5.5. After getting most of it done, I decided to add several spheres to the scene to simulate a falling motion, and the landing point happened to be the vehicle. However, I noticed that after the spheres fell, they completely passed through the car body without any interaction.

I suspect this issue might be related to collision detection, but since I am new to PhysX, I'm having difficulty resolving it on my own. This issue has been troubling me for several days. Any advice or guidance would be greatly appreciated!
## Picture
Attached is a picture from my simulation in PVD, where you can see that the small spheres are passing through the vehicle's body.

[image]

vreutskyy, 2025-02-11:
Hi @capp-adocia 
I can't say what's wrong in you code without seeing it, but as for SnippetVehicleDirectDrive, it intentionally suppresses all collision detection and only uses scene queries. To enable it, you'd need to change the filter shader so it sets PxPairFlag::eDETECT_DISCRETE_CONTACT and PxPairFlag::eSOLVE_CONTACT in pairFlags, and then returns PxFilterFlag::eDEFAULT. Also, you'd need to set PxShapeFlag::eSIMULATION_SHAPE to the vehicle frame shape, so it would be used in collision detection and contact solving.
```
PxFilterFlags VehicleFilterShader(
PxFilterObjectAttributes attributes0, PxFilterData filterData0,
PxFilterObjectAttributes attributes1, PxFilterData filterData1,
PxPairFlags& pairFlags, const void* constantBlock, PxU32 constantBlockSize)
{
	PX_UNUSED(attributes0);
	PX_UNUSED(filterData0);
	PX_UNUSED(attributes1);
	PX_UNUSED(filterData1);
	PX_UNUSED(pairFlags);
	PX_UNUSED(constantBlock);
	PX_UNUSED(constantBlockSize);
	pairFlags = PxPairFlag::eDETECT_DISCRETE_CONTACT | PxPairFlag::eSOLVE_CONTACT; // add this
	return PxFilterFlag::eDEFAULT/*eSUPPRESS*/; // change this
}
```
```
		...
		const PxVehiclePhysXRigidActorShapeParams physxActorShapeParams(boxGeom, physxParams.physxActorBoxShapeLocalPose, defaultMaterial, PxShapeFlags(PxShapeFlag::eSIMULATION_SHAPE/*0*/), PxFilterData(), PxFilterData());
		...
```

[image]

capp-adocia, 2025-02-12:
@vreutskyy  Oh, the code you provided is really effective. I used the files in the SnippetVehicleDirectDrive project, but the difference is that I use hard-coded loading parameters. I only modified the VehicleFilterShader function before, which is almost the same as what you provided. What I didn't expect was that I also needed to modify the physxActorShapeParams parameters in PhysXIntegration.cpp for it to work. This made me realize that my basics are greatly lacking. Thank you very much for your reply. By the way, I also wanted to add another vehicle to the scene before. Is this modification appropriate?
```
PxVehiclePhysXSimulationContext gVehicleSimulationContext;
//The vehicle with engine drivetrain
DirectDriveVehicle gVehicle; // 第一辆车

PxVehiclePhysXSimulationContext gVehicleSimulationContext2;
DirectDriveVehicle gVehicle2;  // 添加第二辆车的对象
```
```
void stepPhysics()
{
	const PxReal timestep = 1.0f / 60.0f;

	// 第一辆车的物理步进
	const PxVec3 linVel1 = gVehicle.mPhysXState.physxActor.rigidBody->getLinearVelocity();
	const PxVec3 forwardDir1 = gVehicle.mPhysXState.physxActor.rigidBody->getGlobalPose().q.getBasisVector2();
	const PxReal forwardSpeed1 = linVel1.dot(forwardDir1);

	const PxU8 nbSubsteps1 = (forwardSpeed1 < 5.0f ? 3 : 1);
	gVehicle.mComponentSequence.setSubsteps(gVehicle.mComponentSequenceSubstepGroupHandle, nbSubsteps1);
	gVehicle.step(timestep, gVehicleSimulationContext);

	// 第二辆车的物理步进
	const PxVec3 linVel2 = gVehicle2.mPhysXState.physxActor.rigidBody->getLinearVelocity();
	const PxVec3 forwardDir2 = gVehicle2.mPhysXState.physxActor.rigidBody->getGlobalPose().q.getBasisVector2();
	const PxReal forwardSpeed2 = linVel2.dot(forwardDir2);

	const PxU8 nbSubsteps2 = (forwardSpeed2 < 5.0f ? 3 : 1);
	gVehicle2.mComponentSequence.setSubsteps(gVehicle2.mComponentSequenceSubstepGroupHandle, nbSubsteps2);
	gVehicle2.step(timestep, gVehicleSimulationContext2);

	// Forward integrate the PhysX scene by a single timestep.
	gScene->simulate(timestep);
	gScene->fetchResults(true);
}
```
```
bool initVehicles()
{
	// 现有的第一辆车的初始化
	LoadBaseParams(gVehicle.mBaseParams);
	setPhysXIntegrationParams(gVehicle.mBaseParams.axleDescription, gPhysXMaterialFrictions, gNbPhysXMaterialFrictions, gPhysXDefaultMaterialFriction, gVehicle.mPhysXParams);
	LoadDirectDrivetrainParams(gVehicle.mBaseParams.axleDescription, gVehicle.mDirectDriveParams);
	if (!gVehicle.initialize(*gPhysics, PxCookingParams(PxTolerancesScale()), *gMaterial))
	{
		return false;
	}

	// 添加第二辆车的初始化过程
	LoadBaseParams(gVehicle2.mBaseParams);  // 为第二辆车加载参数
	setPhysXIntegrationParams(gVehicle2.mBaseParams.axleDescription, gPhysXMaterialFrictions, gNbPhysXMaterialFrictions, gPhysXDefaultMaterialFriction, gVehicle2.mPhysXParams);
	LoadDirectDrivetrainParams(gVehicle2.mBaseParams.axleDescription, gVehicle2.mDirectDriveParams);
	if (!gVehicle2.initialize(*gPhysics, PxCookingParams(PxTolerancesScale()), *gMaterial))
	{
		return false;
	}

	// 设置车辆初始状态
	gVehicle.mTransmissionCommandState.gear = PxVehicleDirectDriveTransmissionCommandState::eFORWARD;
	gVehicle2.mTransmissionCommandState.gear = PxVehicleDirectDriveTransmissionCommandState::eFORWARD;
	 
	// 设置车辆的初始位置和方向
	PxTransform pose1(PxVec3(0.0f, 0.2f, 0.0f), PxQuat(PxIdentity));  // 第一辆车
	PxTransform pose2(PxVec3(2.0f, 0.2f, 0.0f), PxQuat(PxIdentity));  // 第二辆车位置在X轴方向上偏移

	gVehicle.setUpActor(*gScene, pose1, "DirectDriver1");
	gVehicle2.setUpActor(*gScene, pose2, "DirectDriver2");

	// 初始化模拟上下文
	gVehicleSimulationContext.setToDefault();
	gVehicleSimulationContext.frame.lngAxis = PxVehicleAxes::ePosZ;
	gVehicleSimulationContext.frame.latAxis = PxVehicleAxes::ePosX;
	gVehicleSimulationContext.frame.vrtAxis = PxVehicleAxes::ePosY;
	gVehicleSimulationContext.scale.scale = 1.0f;
	gVehicleSimulationContext.gravity = gGravity;
	gVehicleSimulationContext.physxScene = gScene;
	gVehicleSimulationContext.physxActorUpdateMode = PxVehiclePhysXActorUpdateMode::eAPPLY_ACCELERATION;

	// 第二辆车的模拟上下文
	gVehicleSimulationContext2.setToDefault();
	gVehicleSimulationContext2.frame.lngAxis = PxVehicleAxes::ePosZ;
	gVehicleSimulationContext2.frame.latAxis = PxVehicleAxes::ePosX;
	gVehicleSimulationContext2.frame.vrtAxis = PxVehicleAxes::ePosY;
	gVehicleSimulationContext2.scale.scale = 1.0f;
	gVehicleSimulationContext2.gravity = gGravity;
	gVehicleSimulationContext2.physxScene = gScene;
	gVehicleSimulationContext2.physxActorUpdateMode = PxVehiclePhysXActorUpdateMode::eAPPLY_ACCELERATION;

	return true;
}
```

capp-adocia, 2025-02-12:
> Seems okay. I see you've duplicated the initialization and stepping to process the 2nd vehicle. You could also declare an array of vehicles and wrap those places into a loop each to process an arbitrary number of vehicles.

Indeed, what a terrible experience it would be if I didn't use containers. If I have questions about this later, can I post them in **Issues**? Or where can I find a site like a developer community to look for possible solutions?

capp-adocia, 2025-02-12:
I understand, but it's a bit embarrassing. I usually submit questions like this in **Discussions**. I always thought that only **Issues** can post user questions. Thank you for your answer.

closed 2025-02-14 by capp-adocia