NVIDIA-Omniverse/PhysX #369: How to write code for manual transmission vehicles

capp-adocia, 2025-02-19:
## Hi

I use an engine-driven vehicle, not a manual transmission, but an automatic transmission. I thought it would be as easy to use as an automatic transmission, but it still failed. I studied the documentation of this SDK on purpose, but in the end I still couldn't start the vehicle normally. 

when I first pressed the q key to step on the clutch, then pressed the w key to switch the current gear currentGear to 2, then pressed the forward key, and finally released the q key to release the clutch, the car could run, but after running for a while it returned to neutral. I don't know what to do. I think there is still a problem with the code I wrote, so I came here to ask for advice. If I want to realize manual transmission drive of the vehicle, how should I write the code?

Once I have resolved the issue, I will close this issue.

## Code
```cpp
// 设置前进
void throttle()
{
	// 设置油门
	gVehicle.mCommandState.throttle = 0.8f;  // 油门
	gVehicle.mCommandState.brakes[0] = 0.0f;  // 刹车保持关闭
	gVehicle.mCommandState.brakes[1] = 0.0f;  // 刹车保持关闭

	//gVehicle.mTransmissionCommandState.targetGear = PxVehicleEngineDriveTransmissionCommandState::eAUTOMATIC_GEAR;
}
void handleThrottle(int action) {
	if (action == SDL_KEYDOWN) {
		throttle();
	}
	else if (action == SDL_KEYUP) {
		brake();
	}
}
void OnKey(int key, int scancode, int action, int mods) {
	if (action == SDL_KEYDOWN || action == SDL_KEYUP) {
		// 处理油门、后退、转向
		switch (key) {
		case SDLK_KP_8:
			handleThrottle(action); // forward
			break;
		case SDLK_KP_2:
			handleBackward(action);
			break;
		case SDLK_KP_4:
			handleSteering(action, true); // true left
			break;
		case SDLK_KP_6:
			handleSteering(action, false); // false right
			break;
		case SDLK_SPACE:
			createDynamic(PxTransform(PxVec3(0, 20, 2), PxQuat(1.0f, 0.0f, 0.0f, 0.0f)), PxSphereGeometry(0.5f), PxVec3(0., 1., 0.) * 10.0f);
			break;
		case SDLK_LSHIFT:
			VehicleShift(action);
			break;
			// 加速
		case SDLK_LCTRL:
			VehicleAccelerate(action);
			break;
		case SDLK_q:
			if (action == SDL_KEYDOWN && gVehicle.mTransmissionCommandState.clutch == 0.0f) {
				gVehicle.mTransmissionCommandState.clutch = 1.0f;  // clutch
			}
			if (action == SDL_KEYUP && gVehicle.mTransmissionCommandState.clutch == 1.0f) {
				gVehicle.mTransmissionCommandState.clutch = 0.0f;  // release clutch
			}
			break;
		case SDLK_w:
			if (action == SDL_KEYDOWN && gVehicle.mTransmissionCommandState.clutch == 1.0f) {
				gVehicle.mEngineDriveState.gearboxState.currentGear = 2;
			}
			break;
		default:
			break;
		}

		cameraControl->onKey(key, action, mods);
	}
}

```
## Video
https://github.com/user-attachments/assets/ca89bc6c-c585-4a78-a85c-9aac253bd28e

capp-adocia, 2025-03-01:
@vreutskyy After mulling it over these days, I think that's the way to change it, geez, targetGear surprisingly has the ability to set values in addition to `eAUTOMATIC_GEAR`, I don't understand why it's designed that way the `enum` in `PxVehicleEngineDriveTransmissionCommandState` only has ` eAUTOMATIC_GEAR`, why not add the gears as an enum type, that's not clear to me?
```cpp
void OnKey(int key, int /*scancode*/, int action, int mods) {
	if (action == SDL_KEYDOWN || action == SDL_KEYUP) {
		// 处理油门、后退、转向
		switch (key) {
		case SDLK_KP_8:
			handleThrottle(action); // forward
			break;
		case SDLK_KP_2:
#ifdef AUTOGEAR
			handleBackward(action); // backward
#endif
			break;
		case SDLK_KP_4:
			handleSteering(action, true); // true left
			break;
		case SDLK_KP_6:
			handleSteering(action, false); // false right
			break;
		case SDLK_SPACE:
			createDynamic(PxTransform(PxVec3(0, 20, 2), PxQuat(1.0f, 0.0f, 0.0f, 0.0f)), PxSphereGeometry(0.5f), PxVec3(0., 1., 0.) * 10.0f);
			break;
		case SDLK_q:
#ifdef MANUALGEAR
			if (action == SDL_KEYDOWN && gVehicle.mTransmissionCommandState.clutch == 0.0f) {
				gVehicle.mTransmissionCommandState.clutch = 1.0f;  // clutch
			}
			if (action == SDL_KEYUP && gVehicle.mTransmissionCommandState.clutch == 1.0f) {
				gVehicle.mTransmissionCommandState.clutch = 0.0f;  // release clutch
			}
#endif
			break;
		case SDLK_w: // 手动挡 升档
#ifdef MANUALGEAR
			if (action == SDL_KEYDOWN && gVehicle.mTransmissionCommandState.clutch == 1.0f && gVehicle.mEngineDriveState.gearboxState.currentGear != 6) {
				gVehicle.mTransmissionCommandState.targetGear = ++gVehicle.mEngineDriveState.gearboxState.currentGear;
			}
#endif
			break;
		case SDLK_e:
#ifdef MANUALGEAR
			if (action == SDL_KEYDOWN && gVehicle.mTransmissionCommandState.clutch == 1.0f && gVehicle.mEngineDriveState.gearboxState.currentGear != 0) {
				gVehicle.mTransmissionCommandState.targetGear = --gVehicle.mEngineDriveState.gearboxState.currentGear;
			}
#endif
			break;
		default:
			break;
		}
		cameraControl->onKey(key, action, mods);
	}
}
```

gyeomannvidia, 2025-03-04:
In the video, I see the target gear fixed at 1 and the current gear change from 2 to 1.

To change gear immediately, set current gear = target gear = currentGear + 1.

For as long as currentGear and targetGear are different, the vehicle will attempt to change gear to targetGear.

capp-adocia, 2025-03-06:
> 在视频中，我看到目标档位固定为 1，当前档位从 2 变为 1。
> 
> 要立即换档，请设置当前档位 = 目标档位 = currentGear + 1。
> 
> 只要 currentGear 和 targetGear 不同，车辆就会尝试换档到 targetGear。

Thanks for your answer. I have basically achieved the switch between manual and automatic gears.

closed 2025-03-06 by vreutskyy