NVIDIA-Omniverse/PhysX #383: Access Violation In GuCollisionSDF.h when Trianglemesh is small

AgentCodeBanks, 2025-03-20:
Hello, 
I have updated PhysX in our program from 5.4.2 to 5.5.1. Now I have noticed that if I have very small triangle meshes (in the mm range) then I get an Access Violation Exception in the GuCollisionSDF.h file. 

[image]

My scene in which this exception occurs looks like this. 

[image]

I let the upper table run towards the lower one with a constant velocity and both tables are of type Kinematic

The same scenario works with PhysX 5.4.2
I don't know why SDF Collision is calculated at all. Is the SDF calculation activated by default in PhysX 5.5.1?

vreutskyy, 2025-03-20:
Hi @AgentCodeBanks 
Thanks for the report. Would be great if you could give us some way to reproduce the issue. A code snippet or something like that.

AgentCodeBanks, 2025-03-24:
Hey @vreutskyy, 
When I have time I will try to break down our programme so that it reproduces the error. Or would the code snippets of our implementation be enough?  If yes, which parts of the code were important?

Until then, is there a way to switch off the SDF so that this exception cannot occur? 

Kind regards

AgentCodeBanks, 2025-03-25:
We are not currently using GPU acceleration. And we have not actually activated SDF collision detection, so our triangle mesh should not create any SDFs. 

I was able to reproduce the error with the SnippedSDF, the files are attached in the zip folder. The problem is probably due to the small size of the mesh, if I scale it up the exception does not occur

[PhysX5.5.1_AccessViolation.zip](https://github.com/user-attachments/files/19446752/PhysX5.5.1_AccessViolation.zip)

nvtw, 2025-04-02:
I can see that the mesh mesh collision path is entered. Can you try setting sceneDesc.kineKineFilteringMode = PxPairFilteringMode::eKILL; on the PxSceneDesc? Maybe also try sceneDesc.staticKineFilteringMode = PxPairFilteringMode::eKILL;

AgentCodeBanks, 2025-04-07:
Now if the 2 flags are not set to True the exception will not occur. But we need these flags to be True because we need to detect collisions between static and kinematic cubes.
So unfortunately this is not an option for us to set them to False. 

Is this a bug that will be fixed in a future version? 
or is this not a bug? then I would be grateful for a suggestion on how to work around the problem.

nvtw, 2025-04-07:
It probably is a bug. I have a fix but it takes time to get it into the PhysX SDK. The only thing I can provide at the moment is the following patch that might help.
```
From e334ea4eb9ba708fc3659d0668a531baad051429 Mon Sep 17 00:00:00 2001
From: somebody <somebody@nvidia.com>
Date: Wed, 2 Apr 2025 14:24:43 +0200
Subject: [PATCH] Add new filter and return no contacts if no sdf is available
 on either of the meshes

---
 .../geomutils/src/contact/GuContactMeshMesh.cpp     |  7 +++++--
 .../source/simulationcontroller/src/ScFiltering.cpp | 13 +++++++++++++
 2 files changed, 18 insertions(+), 2 deletions(-)

diff --git a/physx/source/geomutils/src/contact/GuContactMeshMesh.cpp b/physx/source/geomutils/src/contact/GuContactMeshMesh.cpp
index 4ea0ccf96e..e03d46fc4b 100644
--- a/physx/source/geomutils/src/contact/GuContactMeshMesh.cpp
+++ b/physx/source/geomutils/src/contact/GuContactMeshMesh.cpp
@@ -255,9 +255,12 @@ bool Gu::contactMeshMesh(GU_CONTACT_METHOD_ARGS)
 
 	const bool geom0HasSdf = geom0.triangleMesh->getSDF() != NULL,
 			   geom1HasSdf = geom1.triangleMesh->getSDF() != NULL;
-	if (!(geom0HasSdf && geom1HasSdf))
+	if (!geom0HasSdf && !geom1HasSdf)
+	{
+		return false;
+	}
+	else if (!(geom0HasSdf && geom1HasSdf))
 	{
-		PX_ASSERT(geom0HasSdf || geom1HasSdf); // require at least one SDF
 		if (geom0HasSdf)
 			nbContacts += sdfMeshCollision(transform0, geom0, transform1, geom1, contactReducer, contactDistance, true);
 		else
diff --git a/physx/source/simulationcontroller/src/ScFiltering.cpp b/physx/source/simulationcontroller/src/ScFiltering.cpp
index c03bff4a56..9401aeba95 100644
--- a/physx/source/simulationcontroller/src/ScFiltering.cpp
+++ b/physx/source/simulationcontroller/src/ScFiltering.cpp
@@ -31,6 +31,7 @@
 #include "ScTriggerInteraction.h"
 #include "ScConstraintCore.h"
 #include "ScArticulationSim.h"
+#include "geometry/PxTriangleMesh.h"
 
 using namespace physx;
 using namespace Sc;
@@ -396,6 +397,18 @@ static bool filterRbCollisionPairShared(	FilterInfo& filterInfo, bool& isNonRigi
 	const PxFilterObjectType::Enum filterType0 = PxGetFilterObjectType(filterAttr0);
 	const PxFilterObjectType::Enum filterType1 = PxGetFilterObjectType(filterAttr1);
 
+	if (!isNonRigid)
+	{
+		if (s0.getCore().getGeometryType() == PxGeometryType::eTRIANGLEMESH &&
+			s1.getCore().getGeometryType() == PxGeometryType::eTRIANGLEMESH)
+		{
+			const PxTriangleMeshGeometry& m0 = static_cast<const PxTriangleMeshGeometry&>(s0.getCore().getGeometry());
+			const PxTriangleMeshGeometry& m1 = static_cast<const PxTriangleMeshGeometry&>(s1.getCore().getGeometry());
+			if (m0.triangleMesh->getSDF() == NULL && m1.triangleMesh->getSDF() == NULL)
+				return createFilterInfo(filterInfo, PxFilterFlag::eKILL);
+		}
+	}
+
 	// PT: For unknown reasons the filtering code was not the same for triggers/refiltered pairs and for regular "shape sim" pairs
 	// out of the BP. The tests on "runAllTests" below capture that. I did not change what the code
 	// was doing, although it might very well be wrong - we might want to run all these tests in both codepaths.
-- 
2.43.0.windows.1
```