NVIDIA-Omniverse/PhysX #399: Is there a way to calculate the shortest distance between two triangle meshes？

wrsjhhe, 2025-04-18:
I set the Contact Offset to a larger value so that the shortest distance between the two Shapes can be fed back during collision feedback. However, if it is an SDF object, there will be obvious lag. Is there a better way to achieve the shortest distance measurement and return the closest point?

qsgy, 2025-04-23:
me want this too

wrsjhhe, 2025-08-19:
> There is currently no direct way to compute the shortest distance between two triangle meshes.
> 
> The closest I can think of is PxMeshQuery::findOverlapTriangleMesh(). There is a flavor of this function that takes an optional distance parameter, and it would return all pairs of triangles from the two meshes that are closer than this distance. This works on all meshes, even without SDFs, but this is a CPU-only codepath.
> 
> You could perhaps use a proper distance query between the bounds of these meshes (e.g. maybe two bounding spheres or boxes) and switch to the above function to refine the results when the bounds intersect.
> 
> I don't know if this would work in your case.

Thank you, I used the findOverlapTriangleMesh method and it can indeed calculate the shortest distance between two meshes. Just the computational efficiency is a bit slow.