NVIDIA-Omniverse/PhysX #400: AABBTree of AABBPruner become corrupted and scene queries failed

josegd-m, 2025-04-30:
Hi,

We have experienced very rare problems with the AABBPruner incremental builder, in very rare cases the AABBTree has become corrupted, causing Scene Queries to not function properly until a new AABBTree is created.

Upon further investigation, we have determined that this occurs when shapes are added or removed between AABBPruner BUILD_NEW_MAPPING and BUILD_FULL_REFIT build steps, likely because the accumulated fixups (AABBPruner mNewTreeFixups) have not been applied.

This behavior has been observed from PhysX version (3.4) to the latest PhysX version (5.6).

vreutskyy, 2025-04-30:
Hi @josegd-m 
Thanks for the report. We'll check it.

josegd-m, 2025-04-30:
Of course,

The setup of PxSceneQueryDesc is very close to the defaults:
```
sceneDesc	{ 
	gravity	{ x=0.00000000 y=-980.000000 z=0.00000000 }	physx::PxVec3
	simulationEventCallback	0x000000007a8ce558 {...}	physx::PxSimulationEventCallback* {CustomSimulationEventCallback}
	contactModifyCallback	0x0000000000000000 {0x0000000000000000}	class physx::PxContactModifyCallback*
	ccdContactModifyCallback	0x0000000000000000 {0x0000000000000000}	class physx::PxCCDContactModifyCallback*
	filterShaderData	0x0000000000000000	const void*
	filterShaderDataSize	0x00000000	physx::PxU32
	filterShader	0x0000000075a486b0 {CustomFilterShader(physx::PxFilterObjectAttributes, physx::PxFilterData, physx::PxFilterObjectAttributes, physx::PxFilterData, physx::PxPairFlags&, const void*, physx::PxU32)}	physx::PxSimulationFilterShader
	filterCallback	0x0000000000000000 {0x0000000000000000}	class physx::PxSimulationFilterCallback*
	kineKineFilteringMode	eDEFAULT (0x00000003)	enum physx::PxPairFilteringMode::Enum
	staticKineFilteringMode	eDEFAULT (0x00000003)	enum physx::PxPairFilteringMode::Enum
	broadPhaseType	eSAP (0x00000000)	enum physx::PxBroadPhaseType::Enum
	broadPhaseCallback	0x0000000000000000 {0x0000000000000000}	class physx::PxBroadPhaseCallback*
	limits	{ maxNbActors=0x00000000 maxNbBodies=0x00000000 maxNbStaticShapes=0x00000000 maxNbDynamicShapes=0x00000000 maxNbAggregates=0x00000000 ... }	physx::PxSceneLimits
	frictionType	ePATCH (0x00000000)	enum physx::PxFrictionType::Enum
	bounceThresholdVelocity	300.000000	physx::PxReal
	frictionOffsetThreshold	4.00000000	physx::PxReal
	ccdMaxSeparation	4.00000000	physx::PxReal
	solverOffsetSlop	0.00000000	physx::PxReal
	flags	{ mBits=0x00000602 }	physx::PxSceneFlags
	cpuDispatcher	0x000000200995b230 { mWorkerThreads=0x000000200995b360 { mQueueEntryPool={ mTaskEntryPool=0x000000200c0ffb70 {...} mTaskEntryPtrPool={...} } ... } ... }	physx::PxCpuDispatcher* {physx::Ext::DefaultCpuDispatcher}
	gpuDispatcher	0x0000000000000000 {0x0000000000000000}	class physx::PxGpuDispatcher*
	staticStructure	eDYNAMIC_AABB_TREE (0x00000001)	enum physx::PxPruningStructureType::Enum
	dynamicStructure	eDYNAMIC_AABB_TREE (0x00000001)	enum physx::PxPruningStructureType::Enum
	dynamicTreeRebuildRateHint	0x00000064	physx::PxU32
	sceneQueryUpdateMode	eBUILD_ENABLED_COMMIT_ENABLED (0x00000000)	enum physx::PxSceneQueryUpdateMode::Enum
	userData	0x0000000000000000	void*
	solverBatchSize	0x00000080	physx::PxU32
	nbContactDataBlocks	0x00000060	physx::PxU32
	maxNbContactDataBlocks	0x00000060	physx::PxU32
	maxBiasCoefficient	3.40282347e+038	physx::PxReal
	contactReportStreamBufferSize	0x00002000	physx::PxU32
	ccdMaxPasses	0x00000001	physx::PxU32
	wakeCounterResetValue	0.39999997	physx::PxReal
	sanityBounds	{ minimum={ x=-4.00000011e+016 y=-4.00000011e+016 z=-4.00000011e+016 } maximum={ x=4.00000011e+016 y=4.00000011e+016 z=4.00000011e+016 } }	physx::PxBounds3
	gpuDynamicsConfig	{ constraintBufferCapacity=0x02000000 contactBufferCapacity=0x01800000 tempBufferCapacity=0x01000000 contactStreamSize=0x00080000 ... }	physx::PxgDynamicsMemoryConfig
	gpuMaxNumPartitions	0x00000008	physx::PxU32
	gpuComputeVersion	0x00000000	physx::PxU32
	tolerancesScale	{ length=100.000000 mass=1000.00000 speed=1000.00000 
}	physx::PxTolerancesScale
```

and yes, changing the type of structure, the rate hint, or build mode could change the reproduction case,
but it is not desirable for us, becuase shapes are added or removed from the scene constantly at runtime.

Thanks for your time.

josegd-m, 2025-04-30:
This is the SceneDesc from version 342, but I can give you the latest version if needed

PierreTerdiman, 2025-04-30:
We can have a look but I am not sure how to proceed here without a better repro. This is not going to repro with the default parameters in recent PhysX versions (we would have seen that already) and we are unlikely to release a new PhysX 3.x build anyway...

josegd-m, 2025-04-30:
Thrat is PxSceneQueryDesc from 560, and the problem with the AABBTree has become corrupt is the same with this version.

physx::PxSceneQueryDesc
{
	staticStructure	eDYNAMIC_AABB_TREE (0x00000001)	physx::PxPruningStructureType::Enum
	dynamicStructure	eDYNAMIC_AABB_TREE (0x00000001)	physx::PxPruningStructureType::Enum
	dynamicTreeRebuildRateHint	0x00000064	unsigned int
	dynamicTreeSecondaryPruner	eINCREMENTAL (0x00000002)	physx::PxDynamicTreeSecondaryPruner::Enum
	staticBVHBuildStrategy	eFAST (0x00000000)	physx::PxBVHBuildStrategy::Enum
	dynamicBVHBuildStrategy	eFAST (0x00000000)	physx::PxBVHBuildStrategy::Enum
	staticNbObjectsPerNode	0x00000004	unsigned int
	dynamicNbObjectsPerNode	0x00000004	unsigned int
	sceneQueryUpdateMode	eBUILD_ENABLED_COMMIT_ENABLED (0x00000000)	physx::PxSceneQueryUpdateMode::Enum
}

PierreTerdiman, 2025-04-30:
As expected I cannot really reproduce this. Here is a modified SnippetHelloWorld that creates new objects each frame and raycasts against the scene to ensure each object is getting hit. I haven't seen a failure so far.

```
#include <ctype.h>
#include "PxPhysicsAPI.h"
#include "../snippetcommon/SnippetPrint.h"
#include "../snippetcommon/SnippetPVD.h"
#include "../snippetutils/SnippetUtils.h"

using namespace physx;

static PxDefaultAllocator		gAllocator;
static PxDefaultErrorCallback	gErrorCallback;
static PxFoundation*			gFoundation = NULL;
static PxPhysics*				gPhysics	= NULL;
static PxDefaultCpuDispatcher*	gDispatcher = NULL;
static PxScene*					gScene		= NULL;
static PxMaterial*				gMaterial	= NULL;
static PxPvd*					gPvd        = NULL;

static PxReal stackZ = 10.0f;

static PxArray<PxRigidDynamic*>*	gObjects = NULL;

static SnippetUtils::BasicRandom	gRnd;

static PxRigidDynamic* createDynamic(const PxTransform& t, const PxGeometry& geometry, const PxVec3& velocity=PxVec3(0))
{
	PxRigidDynamic* dynamic = PxCreateDynamic(*gPhysics, t, geometry, *gMaterial, 10.0f);
	dynamic->setAngularDamping(0.5f);
	dynamic->setLinearVelocity(velocity);
	gScene->addActor(*dynamic);
	return dynamic;
}

void initPhysics(bool)
{
	gFoundation = PxCreateFoundation(PX_PHYSICS_VERSION, gAllocator, gErrorCallback);

	gPvd = PxCreatePvd(*gFoundation);
	PxPvdTransport* transport = PxDefaultPvdSocketTransportCreate(PVD_HOST, 5425, 10);
	gPvd->connect(*transport,PxPvdInstrumentationFlag::eALL);

	gPhysics = PxCreatePhysics(PX_PHYSICS_VERSION, *gFoundation, PxTolerancesScale(), true, gPvd);

	PxSceneDesc sceneDesc(gPhysics->getTolerancesScale());
	sceneDesc.gravity = PxVec3(0.0f, -9.81f, 0.0f);
	gDispatcher = PxDefaultCpuDispatcherCreate(2);
	sceneDesc.cpuDispatcher	= gDispatcher;
	sceneDesc.filterShader	= PxDefaultSimulationFilterShader;
	gScene = gPhysics->createScene(sceneDesc);

	PxPvdSceneClient* pvdClient = gScene->getScenePvdClient();
	if(pvdClient)
	{
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONSTRAINTS, true);
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONTACTS, true);
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_SCENEQUERIES, true);
	}
	gMaterial = gPhysics->createMaterial(0.5f, 0.5f, 0.6f);
}

void stepPhysics(bool)
{
	gScene->simulate(1.0f/60.0f);
	gScene->fetchResults(true);

	if(!gObjects)
		gObjects = new PxArray<PxRigidDynamic*>;

	// Spawn 10 new objects each frame
	for(PxU32 i=0; i<10; i++)
	{
		const PxVec3 v = gRnd.unitRandomPt() * 100.0f;
		PxRigidDynamic* dyna = createDynamic(PxTransform(v), PxSphereGeometry(1.0f), PxVec3(0.0f));
		gObjects->pushBack(dyna);
	}

	// Raycast against each object, make sure we get a hit
	class MyCallback : public PxQueryFilterCallback
	{
	public:
		virtual	PxQueryHitType::Enum preFilter(const PxFilterData&, const PxShape*, const PxRigidActor* actor, PxHitFlags&)	PX_OVERRIDE	PX_FINAL
		{
			// Filter out all objects except expected one
			if(actor == mExpected)
				return PxQueryHitType::eBLOCK;
			return PxQueryHitType::eNONE;
		}

		virtual PxQueryHitType::Enum postFilter(const PxFilterData&, const PxQueryHit&, const PxShape*, const PxRigidActor*)	PX_OVERRIDE	PX_FINAL
		{
			return PxQueryHitType::eNONE;
		}

		PxRigidDynamic*	mExpected;
	};

	const PxQueryFlags filterFlags = PxQueryFlag::eDYNAMIC|PxQueryFlag::ePREFILTER;
	const PxQueryFilterData filterData(PxFilterData(), filterFlags);

	const PxVec3 dir(0.0f, -1.0f, 0.0f);
	const PxU32 size = gObjects->size();
	for(PxU32 i=0; i<size; i++)
	{
		PxRigidDynamic* current = (*gObjects)[i];
		
		const PxVec3 rayOrigin = current->getGlobalPose().p;

		MyCallback preFilter;
		preFilter.mExpected = current;

		PxRaycastBuffer hit;
		if(gScene->raycast(rayOrigin, dir, 100.0f, hit, PxHitFlag::eDEFAULT, filterData, &preFilter))
		{
			PX_ASSERT(hit.hasAnyHits());
			PX_ASSERT(hit.hasBlock);
			PX_ASSERT(hit.block.actor == current);	// Test that we hit expected object
		}
		else
		{
			PX_ASSERT(0);
		}
	}
}
	
void cleanupPhysics(bool)
{
	PX_RELEASE(gScene);
	PX_RELEASE(gDispatcher);
	PX_RELEASE(gPhysics);
	if(gPvd)
	{
		PxPvdTransport* transport = gPvd->getTransport();
		PX_RELEASE(gPvd);
		PX_RELEASE(transport);
	}
	PX_RELEASE(gFoundation);
	
	printf("SnippetHelloWorld done.\n");
}

void keyPress(unsigned char, const PxTransform&)
{
}

int snippetMain(int, const char*const*)
{
#ifdef RENDER_SNIPPET
	extern void renderLoop();
	renderLoop();
#else
	static const PxU32 frameCount = 100;
	initPhysics(false);
	for(PxU32 i=0; i<frameCount; i++)
		stepPhysics(false);
	cleanupPhysics(false);
#endif

	return 0;
}

```

josegd-m, 2025-05-05:
We have managed to build a deterministic repro case based on your snippet code, here it is:

The frames 13 and 25 matches exactly when the AABBPruner transition from BUILD_NEW_MAPPING to BUILD_FULL_REFIT build steps, and deleting shapes corrupts the tree.

`include <ctype.h>
#include "PxPhysicsAPI.h"
#include "../snippetcommon/SnippetPrint.h"
#include "../snippetcommon/SnippetPVD.h"
#include "../snippetutils/SnippetUtils.h"

using namespace physx;

static PxDefaultAllocator		gAllocator;
static PxDefaultErrorCallback	gErrorCallback;
static PxFoundation*			gFoundation = NULL;
static PxPhysics*				gPhysics	= NULL;
static PxDefaultCpuDispatcher*	gDispatcher = NULL;
static PxScene*					gScene		= NULL;
static PxMaterial*				gMaterial	= NULL;
static PxPvd*					gPvd        = NULL;

static PxReal stackZ = 10.0f;

static PxArray<PxRigidStatic*>*	gObjects = NULL;

static SnippetUtils::BasicRandom	gRnd;

/*
static PxRigidDynamic* createDynamic(const PxTransform& t, const PxGeometry& geometry, const PxVec3& velocity=PxVec3(0))
{
	PxRigidDynamic* dynamic = PxCreateDynamic(*gPhysics, t, geometry, *gMaterial, 10.0f);
	dynamic->setAngularDamping(0.5f);
	dynamic->setLinearVelocity(velocity);
	gScene->addActor(*dynamic);
	return dynamic;
}
*/

static PxRigidStatic* createStatic(const PxTransform& t, const PxGeometry& geometry)
{
	PxRigidStatic* actor = PxCreateStatic(*gPhysics, t, geometry, *gMaterial);
	gScene->addActor(*actor);
	return actor;
}

void initPhysics(bool)
{
	gFoundation = PxCreateFoundation(PX_PHYSICS_VERSION, gAllocator, gErrorCallback);

	gPvd = PxCreatePvd(*gFoundation);
	PxPvdTransport* transport = PxDefaultPvdSocketTransportCreate(PVD_HOST, 5425, 10);
	gPvd->connect(*transport,PxPvdInstrumentationFlag::eALL);

	gPhysics = PxCreatePhysics(PX_PHYSICS_VERSION, *gFoundation, PxTolerancesScale(), true, gPvd);

	PxSceneDesc sceneDesc(gPhysics->getTolerancesScale());
	sceneDesc.gravity = PxVec3(0.0f, -9.81f, 0.0f);
	gDispatcher = PxDefaultCpuDispatcherCreate(2);
	sceneDesc.cpuDispatcher	= gDispatcher;
	sceneDesc.filterShader	= PxDefaultSimulationFilterShader;
	sceneDesc.dynamicTreeRebuildRateHint = 10;
	gScene = gPhysics->createScene(sceneDesc);

	PxPvdSceneClient* pvdClient = gScene->getScenePvdClient();
	if(pvdClient)
	{
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONSTRAINTS, true);
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONTACTS, true);
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_SCENEQUERIES, true);
	}
	gMaterial = gPhysics->createMaterial(0.5f, 0.5f, 0.6f);
}

__declspec(dllimport) bool _Px_BuildStatus;

void stepPhysics(bool)
{
	gScene->simulate(1.0f/60.0f);
	gScene->fetchResults(true);

	if(!gObjects)
	{
		gObjects = new PxArray<PxRigidStatic*>;
		// Spawn 1000 new objects
		for(PxU32 i=0; i<1000; i++)
		{
			const PxVec3 v = gRnd.unitRandomPt() * 1000.0f;
			float s = rand()/(static_cast<PxF32>(0xffff));
			PxRigidStatic* actor = createStatic(PxTransform(v), PxSphereGeometry(s*100.0f) );
			gObjects->pushBack(actor);
		}
	}

	static int frame = 0;
	frame++;

	//if (_Px_BuildStatus)
	if (frame==13 || frame==25)
	{
		// Remove 10 objects
		for(PxU32 i=0; i<10; i++)
		{
			int n = rand()%gObjects->size();
			printf("%d remove %d\n",frame, n);
			PxRigidStatic* actor = (*gObjects)[n];
			gScene->removeActor(*actor);
			gObjects->replaceWithLast(n);
		}
	}

	if (1)
	{
		// Add 1 objects
		for(PxU32 i=0; i<1; i++)
		{
			printf("%d add\n",frame);
			const PxVec3 v = gRnd.unitRandomPt() * 100.0f;
			PxRigidStatic* actor = createStatic(PxTransform(v), PxSphereGeometry(1.0f));	
			gObjects->pushBack(actor);
		}
	}

	// Raycast against each object, make sure we get a hit
	class MyCallback : public PxQueryFilterCallback
	{
	public:
		virtual	PxQueryHitType::Enum preFilter(const PxFilterData&, const PxShape*, const PxRigidActor* actor, PxHitFlags&)	PX_OVERRIDE	PX_FINAL
		{
			// Filter out all objects except expected one
			if(actor == mExpected)
				return PxQueryHitType::eBLOCK;
			return PxQueryHitType::eNONE;
		}

		virtual PxQueryHitType::Enum postFilter(const PxFilterData&, const PxQueryHit&, const PxShape*, const PxRigidActor*)	PX_OVERRIDE	PX_FINAL
		{
			return PxQueryHitType::eNONE;
		}

		PxRigidStatic*	mExpected;
	};

	const PxQueryFlags filterFlags = PxQueryFlag::eSTATIC|PxQueryFlag::ePREFILTER;
	const PxQueryFilterData filterData(PxFilterData(), filterFlags);

	const PxVec3 dir(0.0f, -1.0f, 0.0f);
	const PxU32 size = gObjects->size();
	for(PxU32 i=0; i<size; i++)
	{
		PxRigidStatic* current = (*gObjects)[i];
		
		const PxVec3 rayOrigin = current->getGlobalPose().p;

		MyCallback preFilter;
		preFilter.mExpected = current;

		PxRaycastBuffer hit;
		if(gScene->raycast(rayOrigin, dir, 0.1f, hit, PxHitFlag::eDEFAULT, filterData, &preFilter))
		{
			PX_ASSERT(hit.hasAnyHits());
			PX_ASSERT(hit.hasBlock);
			PX_ASSERT(hit.block.actor == current);	// Test that we hit expected object
		}
		else
		{
			PX_ASSERT(0);
		}
	}
}
	
void cleanupPhysics(bool)
{
	PX_RELEASE(gScene);
	PX_RELEASE(gDispatcher);
	PX_RELEASE(gPhysics);
	if(gPvd)
	{
		PxPvdTransport* transport = gPvd->getTransport();
		PX_RELEASE(gPvd);
		PX_RELEASE(transport);
	}
	PX_RELEASE(gFoundation);
	
	printf("SnippetHelloWorld done.\n");
}

void keyPress(unsigned char, const PxTransform&)
{
}

int snippetMain(int, const char*const*)
{
#ifdef RENDER_SNIPPET
	extern void renderLoop();
	renderLoop();
#else
	static const PxU32 frameCount = 100;
	initPhysics(false);
	for(PxU32 i=0; i<frameCount; i++)
		stepPhysics(false);
	cleanupPhysics(false);
#endif

	return 0;
}`

PierreTerdiman, 2025-05-05:
Thank you, this is very useful. I am able to reproduce the problem, although I haven't found the bug yet. This may take a while.

josegd-m, 2025-05-06:
Thank you very much for the fix, we'll give a try and come later with our feedback.

Yeah, it's very weird that this bug haven't seen before, it's not very easy to repro but not imposible, and more probable in the context of a game with a streamed world.

josegd-m, 2025-05-08:
Hello again, we were unable to reproduce the problem with your tentative fix. 
This sounds like it could be a solution.

Previously, we have tried our own solution, which solved the problem also,
consisting of apply accumulated fixups at mProgress==BUILD_FULL_REFIT of previously AABBPruner::buildstep, like this:

```
		else if(mProgress==BUILD_FULL_REFIT)
		{
			mNbCalls++;
			mProgress = BUILD_LAST_FRAME;

			// BugFix?
			if(mNewTreeFixups.size())
			{
				mNewTreeMap.initMap(PxMax(mPool.getNbActiveObjects(), mNbCachedBoxes), *mNewTree);

				// The new mapping has been computed using only indices stored in the new tree. Those indices map the pruning pool
				// we had when starting to build the tree. We need to re-apply recorded moves to fix the tree.
				for(NewTreeFixup* r = mNewTreeFixups.begin(); r < mNewTreeFixups.end(); r++)
					mNewTreeMap.invalidate(r->removedIndex, r->relocatedLastIndex, *mNewTree);

				mNewTreeFixups.clear();
			}

			{
				PX_PROFILE_ZONE("SceneQuery.prunerNewTreeFullRefit", mContextID);

				// We need to refit the new tree because objects may have moved while we were building it.
				mNewTree->fullRefit(mPool.getCurrentWorldBoxes());
			}
		}
```

Anyway, we're not sure which one be the right solution.

PierreTerdiman, 2025-05-08:
I would have to dig into this and analyze your version to be entirely sure. Gut feeling is that my tentative fix feels better, as it happens during BUILD_FINISHED at the very last moment. Yours happens during BUILD_FULL_REFIT, and there is still BUILD_LAST_FRAME after that. It could be that the same problem happens again during that last frame. I am not sure, as this code is quite complicated now.

liu-free, 2026-09-08:
> This has been fixed. Fix will appear in 5.11.0.

Hello, I seem to be encountering the same problem. Could you tell me what the official fix code looks like?