NVIDIA-Omniverse/PhysX #402: PhysX CUDA Context Manager binds successfully but cuModule load fails

ChenlizheMe, 2025-05-06:
### Library and Version

PhysX v5.3.1

### Operating System

Ubuntu 20.04 LTS

### Steps to Trigger Behavior

 1. Build and load a standalone shared object (.so) that uses the CUDA Driver API to load a module—this succeeds.
 2. In the same process, call our helper PX_SelectGPU(_Foundation, 0) to bind PhysX to the pre-existing CUcontext.
 3. During PxCreateCudaContextManager, PhysX attempts to load a CUDA module but fails (error code 200), even though the PxCudaContextManager is successfully created and reports a valid context.

### Code Snippet to Reproduce Behavior

```cpp
physx::PxCudaContextManager* PX_SelectGPU(physx::PxFoundation* _Foundation,
                                          int gpuid) {
    dexsim::compute::DFComputeCore::Initialize(true);
    auto cumgr = dexsim::compute::DFComputeCore::Instance()
                    .GetCudaMgr()->GetCuda();
    CUcontext cuContext;
    CUdevice  device;
    
    // Select device and get current context
    cumgr->cuDeviceGet(&device, gpuid);
    cumgr->cuCtxGetCurrent(&cuContext);

    // Log compute capability
    int major = 0, minor = 0;
    cumgr->cuDeviceGetAttribute(&major, CU_DEVICE_ATTRIBUTE_COMPUTE_CAPABILITY_MAJOR, device);
    cumgr->cuDeviceGetAttribute(&minor, CU_DEVICE_ATTRIBUTE_COMPUTE_CAPABILITY_MINOR, device);
    utility::LogInfo("Compute Capability: sm_{}{}\n", major, minor);

    // Bind PhysX to the existing context
    physx::PxCudaContextManagerDesc desc;
    desc.ctx = &cuContext;
    auto cudaMgr = PxCreateCudaContextManager(*_Foundation, desc,
                                              PxGetProfilerCallback());
    if (!cudaMgr || !cudaMgr->contextIsValid()) {
        if (cudaMgr) cudaMgr->release();
        _Foundation->error(physx::PxErrorCode::eINVALID_OPERATION,
                          __FILE__, __LINE__,
                          "Failed to initialize PhysX CUDA Context Manager");
        return nullptr;
    }

    utility::LogInfo(
        "PhysX successfully bound to existing CUDA context on GPU {}",
        gpuid);

    return cudaMgr;
}

```

###  Behavior

/buildAgent/work/eb2f45c4acc808a0/physx/source/cudamanager/src/CudaContextManager.cpp (778) : internal error : Failed to load CUDA module data. Cuda error code 200.

[DexSim WARNING] ERROR:32,Failed to load CUDA module data. Cuda error code 200.
,/buildAgent/work/eb2f45c4acc808a0/physx/source/cudamanager/src/CudaContextManager.cpp,778
[DexSim INFO] /home/dex/project/dexsim/cpp/dexsim/engine/adaptation/physx/DFPhysicAdapt.cpp:116: PhysX successfully bound to existing CUDA context on GPU 0

ChenlizheMe, 2025-05-06:
Inside the same .so file, loading the required .ptx file manually via the CUDA Driver API succeeds; the .so links to CUDA using dlopen with the following code:
```cpp
void* cuda_lib = dlopen("/usr/lib/x86_64-linux-gnu/libcuda.so", RTLD_NOW | RTLD_GLOBAL);
```

ialgniq1974, 2025-05-08:
The error could be related to CUDA lazy loading. To disable CUDA lazy loading, set the environment variable CUDA_MODULE_LOADING to EAGER before launching your application. For example:

`export CUDA_MODULE_LOADING=EAGER`

ChenlizheMe, 2025-05-08:
> The error could be related to CUDA lazy loading. To disable CUDA lazy loading, set the environment variable CUDA_MODULE_LOADING to EAGER before launching your application. For example:
> 
> `export CUDA_MODULE_LOADING=EAGER`

Thank you for the suggestion! I tried setting CUDA_MODULE_LOADING=EAGER and then recompiled the code, but unfortunately I’m still getting the same error with code 200.

Just to clarify, I’m using libcuda.so from the path /usr/lib/x86_64-linux-gnu/libcuda.so.

Let me know if there’s anything else I could try—really appreciate your help!

ialgniq1974, 2025-05-09:
Can you try changing libcuda.so to libcuda.so.1

ChenlizheMe, 2025-05-09:
> Can you try changing libcuda.so to libcuda.so.1

Thanks for replying, I've just tried this, but it also doesn't work.

ialgniq1974, 2025-05-09:
Are you able to provide a simple repro?

ChenlizheMe, 2025-05-09:
Hi @ialgniq1974,

I’ve extracted a **minimal, self‑contained repro** and pushed it here:
&lt;https://github.com/ChenlizheMe/PhysX-Warp&gt;

### Build &amp; run

```bash
# clone (submodules not needed, but just in case)
git clone --recursive https://github.com/ChenlizheMe/PhysX-Warp.git
cd PhysX-Warp

# 1) build the tiny CUDA helper library
cd cuda
mkdir build &amp;&amp; cd build
cmake ..
make -j              # produces libwarp_cuda.so
cd ../..

# 2) build the main demo that links PhysX + the helper
mkdir build &amp;&amp; cd build
cmake ..
make -j              # produces ./main

# 3) run – the error still appears
./main
```

</code></pre>
<p>On my machine the output is identical to the original report:</p>
<pre><code>internal error : Failed to load CUDA module data. Cuda error code 200.
</code></pre>
<h3>Environment</h3>

item | value
-- | --
OS | Ubuntu 20.04 LTS
GPU driver | 550.54.15
CUDA toolkit | 12.1 (nvcc 12.1.66)
PhysX | v5.3.1 (tag)
Compiler | GCC 9.4.0, clang 10.0.1

<p>The repo contains just two <code inline="">.cpp</code> files: one that loads a small PTX via the CUDA Driver API (works) and another that binds PhysX to the same <code inline="">CUcontext</code>, reproducing the failure.</p>
<p>Please let me know if any additional logs, configs, or build flags would be useful—I’m happy to collect them. Thanks again for your time!</p>

DW01, 2025-06-06:
I wrote this while half-asleep, and clearly didn't go back and verify what I'd written.

There's a lot of crap here that I'd mistakenly assumed was universally true, but as it turns out, the key point I tried to make is only somewhat true of 32-bit platforms, which is not only irrelevant for modern development, it also only applies to games released before around 2009 anyway (it's **32-bit** PhysX that got deprecated. D'oh!). That's my bad, and throws the whole issue into question.

That being said, I still think attempting (my own C++ knowledge isn't amazing, so it'll be a lot of learning as I go) to write an engine plugin for Jolt is a decent idea, largely from the perspective of both choice and understanding what's possible. I can't make any guarantees, but the API is well-documented and the native PhysX integration itself can serve as a reference on how to do things, so I'll probably take this up in my spare time. :)

Once again, my bad for making outlandish assumptions while half-asleep!

ChenlizheMe, 2025-06-06:
Thanks for the follow-up — and no worries at all, we’ve all been there!

I actually managed to resolve the issue on my end. The problem turned out to be related to how I was using the CUDA Driver API. Instead of including `cuda.h` directly, I was loading the DLL manually, which seems to cause some incompatibility with PhysX.

After switching to directly including `cuda.h`, everything started working as expected. I'll go ahead and close this issue now. Thanks again!