NVIDIA-Omniverse/PhysX #407: Simulation with TGS can blow up and create NaNs #2

ikkah, 2025-05-23:
Hi.

I previously reported an issue (https://github.com/NVIDIA-Omniverse/PhysX/issues/340) which was closed as I was using arbitrary values to set the mass space inertia tensor. I changed the code to use physx::PxRigidBodyExt::updateMassAndInertia() as was instructed and the issue still reproduces. I'm creating a new ticket as I can't reopen the previous ticket or comment on it.

### Library and Version

v5.4.2-106.1

### Operating System

Windows 11

### Steps to Trigger Behavior

Run the attached test code in place of SnippetContactReport.cpp in the snippets framework.

### Code Snippet to Reproduce Behavior

```
// Redistribution and use in source and binary forms, with or without
// modification, are permitted provided that the following conditions
// are met:
//  * Redistributions of source code must retain the above copyright
//    notice, this list of conditions and the following disclaimer.
//  * Redistributions in binary form must reproduce the above copyright
//    notice, this list of conditions and the following disclaimer in the
//    documentation and/or other materials provided with the distribution.
//  * Neither the name of NVIDIA CORPORATION nor the names of its
//    contributors may be used to endorse or promote products derived
//    from this software without specific prior written permission.
//
// THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS ''AS IS'' AND ANY
// EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
// IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR
// PURPOSE ARE DISCLAIMED.  IN NO EVENT SHALL THE COPYRIGHT OWNER OR
// CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL,
// EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO,
// PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR
// PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY
// OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
// (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
// OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
//
// Copyright (c) 2008-2024 NVIDIA Corporation. All rights reserved.
// Copyright (c) 2004-2008 AGEIA Technologies, Inc. All rights reserved.
// Copyright (c) 2001-2004 NovodeX AG. All rights reserved.  

// ****************************************************************************
// This snippet illustrates the use of simple contact reports.
//
// It defines a filter shader function that requests touch reports for 
// all pairs, and a contact callback function that saves the contact points.  
// It configures the scene to use this filter and callback, and prints the 
// number of contact reports each frame. If rendering, it renders each 
// contact as a line whose length and direction are defined by the contact 
// impulse.
// 
// ****************************************************************************

#include <vector>
#include "PxPhysicsAPI.h"
#include "../snippetutils/SnippetUtils.h"
#include "../snippetcommon/SnippetPrint.h"
#include "../snippetcommon/SnippetPVD.h"

using namespace physx;

static PxDefaultAllocator		gAllocator;
static PxDefaultErrorCallback	gErrorCallback;
static PxFoundation* gFoundation = NULL;
[... 213 lines cut ...]
vel: 0.0064527 - pos -0.00160286
vel: 0.010192 - pos -0.00290192
vel: 0.0163399 - pos -0.00444228
vel: 0.510571 - pos -0.00611298
vel: 0.328636 - pos -0.00784011
vel: 0.630298 - pos -0.00967739
vel: 0.629773 - pos -0.0115253
vel: 83.4777 - pos -0.013018
vel: 83.4081 - pos -0.0110605
vel: 83.5819 - pos -0.00186123
vel: 83.5123 - pos 0.00733888
vel: 89.448 - pos -0.105777
vel: 89.2476 - pos -0.216657
vel: 88.8806 - pos -0.325738
vel: 88.046 - pos -0.432816
vel: 87.9726 - pos 0.800082
vel: 63.6873 - pos 5.96276
vel: 63.6343 - pos -82.9144
vel: 63.5812 - pos -452.974
vel: -nan(ind) - pos 4.13055e+07
```

Running the code in the debug config also triggers some internal asserts in DyTGSDynamics.cpp.

Commenting out the line that enables physx::PxSolverType::eTGS fixes the issue.

vreutskyy, 2025-05-23:
Thanks @ikkah, I'll check this sample too

ikkah, 2025-05-23:
Hi @vreutskyy, thank you for investigating the problem.

I agree that the code in stepPhysics() looks rather interesting. I was able to simplify the snippet a little more and removed the code added to stepPhysics() entirely. I've also made the root rigid body static. I'll add the entire file below for clarity.

The velocity (which I've reduced a little bit down to 100 m/s) is rather high but is something we encounter regularly, especially for rigid bodies affected by gravity.

The setup in the snippet is a simplified version of a real problem we have in our game. The joint limits are the values from a ragdoll setup created by one of our rigging artists. The linear limit doesn't seem too suspicious to me, except maybe for the short extent, but the code doesn't seem to be sensitive to the exact value of that (replacing 0.05f to 0.5f still reproduces the bug).

```
// Redistribution and use in source and binary forms, with or without
// modification, are permitted provided that the following conditions
// are met:
//  * Redistributions of source code must retain the above copyright
//    notice, this list of conditions and the following disclaimer.
//  * Redistributions in binary form must reproduce the above copyright
//    notice, this list of conditions and the following disclaimer in the
//    documentation and/or other materials provided with the distribution.
//  * Neither the name of NVIDIA CORPORATION nor the names of its
//    contributors may be used to endorse or promote products derived
//    from this software without specific prior written permission.
//
// THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS ''AS IS'' AND ANY
// EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
// IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR
// PURPOSE ARE DISCLAIMED.  IN NO EVENT SHALL THE COPYRIGHT OWNER OR
// CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL,
// EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO,
// PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR
// PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY
// OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
// (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
// OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
//
// Copyright (c) 2008-2024 NVIDIA Corporation. All rights reserved.
// Copyright (c) 2004-2008 AGEIA Technologies, Inc. All rights reserved.
// Copyright (c) 2001-2004 NovodeX AG. All rights reserved.  

// ****************************************************************************
// This snippet illustrates the use of simple contact reports.
//
// It defines a filter shader function that requests touch reports for 
// all pairs, and a contact callback function that saves the contact points.  
// It configures the scene to use this filter and callback, and prints the 
// number of contact reports each frame. If rendering, it renders each 
// contact as a line whose length and direction are defined by the contact 
// impulse.
// 
// ****************************************************************************

#include <vector>
#include "PxPhysicsAPI.h"
#include "../snippetutils/SnippetUtils.h"
#include "../snippetcommon/SnippetPrint.h"
#include "../snippetcommon/SnippetPVD.h"

using namespace physx;

static PxDefaultAllocator		gAllocator;
static PxDefaultErrorCallback	gErrorCallback;
static PxFoundation* gFoundation = NULL;
static PxPhysics* gPhysics = NULL;
static PxDefaultCpuDispatcher* gDispatcher = NULL;
static PxScene* gScene = NULL;
static PxPvd* gPvd = NULL;

std::vector<PxVec3> gContactPositions;
std::vector<PxVec3> gContactImpulses;

static PxFilterFlags contactReportFilterShader( PxFilterObjectAttributes attributes0, PxFilterData filterData0,
	PxFilterObjectAttributes attributes1, PxFilterData filterData1,
[... 142 lines cut ...]
	{
		PxPvdTransport* transport = gPvd->getTransport();
		PX_RELEASE( gPvd );
		PX_RELEASE( transport );
	}
	PX_RELEASE( gFoundation );

	printf( "SnippetContactReport done.\n" );
}

int snippetMain( int, const char* const* )
{
#ifdef RENDER_SNIPPET
	extern void renderLoop();
	renderLoop();
#else
	initPhysics( false );
	for( PxU32 i = 0; i < 250; i++ )
		stepPhysics( false );
	cleanupPhysics( false );
#endif

	return 0;
}
```