NVIDIA-Omniverse/PhysX #425: Invalid PxMemCopy in Scene::preallocateContactManagers

unlightable, 2025-07-25:
### Library and Version

PhysX v5.6.0

### Operating System

Centos 7

### Steps to Trigger Behavior
More than `OverlapFilterTask::MaxPairs` (64 right now) overlaps should be created by broadphase and some of them should be filtered during `NPhaseCore::runOverlapFilters` e.g. by returning `PxFilterFlag::eKILL` from filter shader.

In `Sc::Scene::preallocateContactManagers` filtered overlaps are being gathered/compacted inside `AABBManagerBase::mCreatedOverlaps`.
It is incorrectly done with `PxMemCopy` instead of `PxMemMove`, since `pairs`/`OverlapFilterTask::mPairs` are spans in `mCreatedOverlaps` and therefore may overlap.

### Expected Behavior

No overlaps are lost/duplicated.

### Actual Behavior

Due to different `memcpy` implementations handling overlapping regions differently, actual behavior is undefined.
Issue happens for me on very old libc provided by Centos 7: it fills `dest` with multiple copies of ~64byte sized region, or something like that.
It is still incorrect to `memcpy` overlapping regions though!

vreutskyy, 2025-07-25:
Hi @unlightable 
Thanks for the report. We'll check it.

Morten-Hilker-Skaaning, 2025-09-25:
We hit this in 5.6.1 5ca9f47 (5ca9f472105a90d70d957c243cb0ef36fe251a9f).
1488 bytes overlap.